floripasat/obdh #46: Send Internal sensors data to the store data task

eldertramontin, 2017-03-09:
Send temperature, voltage and current through a queue.

mariobaldini, 2017-03-23:
Já existe uma task que lê periodicamente valores de sensores (ex: temperatura, corrente, etc..) e salva na memória nv?

eldertramontin, 2017-03-23:
Sim. Uma task lê os sensores e envia os dados a outra task que salva na memória não volátil.

mariobaldini, 2017-03-24:
Onde está documentado o formato de como esses dados são salvos na memória? (e como extrair/decodificar eles)

eldertramontin, 2017-03-27:
Não está. Vou verificar onde é melhor para documentar isso, criar uma issue referente e linkar assim que o fizer.

mariobaldini, 2017-03-29:
Blz.

Mas a documentação sempre tem q fazer parte da atividade em si. Organização e documentação do projeto tem mais importância do que código fonte em si (do primeiro é mais fácil chegar no segundo; do que o contrario).

Quando resolver uma issue, está implicito que o resultado foi propriamente documentado. Caso não tiver sido feito ainda, pode deixar pendente na propria issue (e não fechá-la ainda).

eldertramontin, 2017-04-04:
https://github.com/floripasat/obdh/blob/master/firmware/docs/packet_description.md#msp-sensors

closed 2017-04-04 by eldertramontin