floripasat/obdh #57: Do Solar panels self-test on task start

eldertramontin, 2017-03-15:
Read some test registers to verify if the solar panels are working.

mariobaldini, 2017-03-16:
Não tem como fazermos um "teste do painel solar". Pra começar q não temos nem interface direta com ele... só o EPS. 

Acho que você quis dizer testar os sensores embarcados na placa do painel solar. (ie. iluminação, etc)
Mas isto é valido para o painel da gomspace... outra marca pode nem ter este sensor. 

OBDH é um modulo por si só, genérico. Não pode ser projetado para uso restrito a um fabricante.

eldertramontin, 2017-03-16:
> "Acho que você quis dizer testar os sensores embarcados na placa do painel solar."

Isso. Realmente, eu quis dizer testar os sensores deles.

> "Pra começar q não temos nem interface direta com ele... só o EPS."

Temos interface direta com eles. USCI_B1, configurada como SPI, comunica com 3 paineis solares (um de cada eixo). Ela, atualmente, só serve para lermos a temperatura dos painéis. Além da SPI, também temos uma entrada para os _sun sensors_, como você falou, e uma saída para o _magnetorquer_ para cada um destes 3 paineis.

mariobaldini, 2017-03-16:
Não temos interface com eles. "Painel solar" é isto: https://solaerotech.com/wp-content/uploads/2015/03/Space-Solar-Cell.jpg
(na verdade é uma célula solar, sendo mais criterioso). 
A saída é uma tensão. Só o EPS tem contato come eles. "Painel/celula solar" não tem SPI, etc..

Temos interface com um termometro, sun sensor, etc... que estão *localizados* no módulo do painel solar. 

O "modulo do painel solar" é que seria a placa que contém as celulas, termometros, encaixes, etc... Os termos acabam ficando meio ambíguos mesmo (até a gomspace usa nomenclaturas meio estranhas). 

É um detalhe meio bobo, mas conforme pudermos ser mais exatos nessas terminologias, economiza tempo no longo prazo (com erros... +5 minutos revisando oq é oq, etc). 

PS: "detalhe bobo" não quer dizer q seja menos importante. Segue um clássico da eng de sw: 
https://raygun.com/blog/2014/05/10-costly-software-errors-history/

NASA’s Mars Climate Orbiter (...) the engineering team failed to make a simple conversion from English units to metric.

mfrata, 2017-03-23:
https://rgsenergy.com/the-difference-between-solar-cells-and-solar-panels/

> 
>The photovoltaic system
>
>Solar panel systems are referred to as photovoltaic systems in the solar industry. [...]
>Solar panels are a combination of multiple solar cells. These cells are arranged in such a way that the solar >panels can capture and combine the electrical output of each cell and send it along a specified path. [...]

closed 2019-04-11 by andrempmattos