floripasat/obdh #63: CCS v7.1 7.1.0.00015 incompatibility

mariobaldini, 2017-03-16:
Error while importing project. 

``
Error: Import failed for project 'obdh' because its compiler definition is not available. Please install the MSP430v16.6 compiler before importing this project - click 'View > CCS App Center' to check if compiler updates are available, or install the compiler manually and register it with CCS through 'Preferences > CCS > Build > Compilers'.
``

eldertramontin, 2017-03-16:
A versão usada do CCS é a 6.1.3.00034. Está sendo usada esta versão para evitar problemas com diferentes versões. Podemos migrar todo o código para uma mais atual, porém será necessário refazer a validação de todo o software.

Para configurar a toolchain, siga este tutorial:
https://github.com/floripasat/obdh/blob/master/firmware/docs/toolchain_setup.txt

mariobaldini, 2017-03-16:
Já estou migrando aqui. (falta alguns testes), mas a principio não teve erro de compatibilidade. 

Faz mais sentido já migrar pro CCSv7, especialmente pq o novo compilador da TI tem bugfixed e é uma versão LTS. TI v16.9.1 LTS

A versão anterior não era LTS, então é melhor fazer toda a validação em uma long term

commit 5f676e53 (2017-03-16): Update project to CCSv7.1 and compiler to LTS: TI v16.9.1 LTS. Closes #63
closed 2017-03-16 by mariobaldini