floripasat/obdh #64: Fix Clocks setup

eldertramontin, 2017-03-16:
The wait time until the clock be stable is short. It is necessary add a larger delay (few seconds).

mariobaldini, 2017-03-16:
Poderia explicar um pouco mais?

Sobre o titulo, este ai está mais com cara de git commit, pois ele 'corrige' o erro. Para issues fica mais legível se for o erro em si (pode omitir que 'é necessário corrigir'). Isso vai estar implicito por uma label de 'bug', improvement, etc.

mariobaldini, 2017-03-16:
Mudei pra um exemplo ali do que acho q entendi. 

Aquele 'too long' ali não está bom, pq é subjetivo e não especifico. O melhor seria ter algo mais objetivo, se viável

eldertramontin, 2017-03-17:
Modifiquei o título e a descrição. Espero que esteja mais claro. Basicamente é preciso só colocar um delay no codigo. Só criei a issue pra justificar a modificação.

closed 2017-03-22 by eldertramontin