floripasat/obdh #65: Verify the need of set the MSP Power Management Module

eldertramontin, 2017-03-22:
Para utilizar clocks elevados (acima de 8MHz) parece ser necessário configurar a tensão do core (Vcore). Verificar se há essa necessidade, visto que ainda não houve nenhum problema relacionado a isto.

mariobaldini, 2017-03-23:
@mfrata o clock do MSP já não estava maior que 8MHz? 

Aquele vez q começamos a usar o cristal externo, não foi para poder deixar o MSP operando em clocks maiores que 8?

mfrata, 2017-03-23:
Isso é para clock interno redundante. Se o cristal falhar ficaremos com o clock padrão de 1,23 MHz. Portanto seria interessante configurar o clock interno para 16 MHz (assim como o externo), para caso ocorra falha apenas perderemos em precisão de velocidade nas execuções. 

Apesar da importância, acho que isso tem que ser feito depois, de termos o minimo rodando

eldertramontin, 2017-03-23:
Na verdade essa issue é para verificar a necessidade de aumentar a tensão do core interno (Vcore). No datasheet diz que para clocks maiores que 8MHz isto faz-se necessário. Essa issue não chega a ser prioridade pois não temos nenhum problema em usar 16MHz, mesmo sem configurar o PMM conforme diz no datasheet. Então ela será tratada futuramente, se necessário for. 

@mfrata, conforme seu comentário anterior, vou criar outra issue para tratar da configuração do oscilador interno (DCO), de maneira a ele ficar configurado a 16MHz também.

mariobaldini, 2017-03-24:
A idéia é o OBDH sempre rodar a 16 mhz; 
a principio baseado no cristal externo; 
que caso tiver algum problema vai fazer um fallback automatico pro interno
que também estará a 16 mhz (mas teria uma precisão um pouco menor)
?

eldertramontin, 2017-03-27:
Exatamente isso.