floripasat/obdh #67: Grants a clock redundancy

eldertramontin, 2017-03-23:
Makes the internal oscillator works on the same frequency of external clock source, so in a failure of the external crystal the uC clock will be approximately the same.

mariobaldini, 2017-03-24:
É possível detectar qual fonte de clock está sendo usada? (interno ou externo).

Se sim, coloca este parametro na lista dos 'dados internos' q são lidos e armazenados periodicamente.

eldertramontin, 2017-03-27:
É possível detectar se houve falha em alguma fonte de clock(cristal 1, cristal 2 e DCO), por meio de _fault flags_ (há uma para cada fonte de clock). Acho que poderia haver uma task para verificar estas e outras possíveis falhas. Acho que o termo é _housekeeping_ se não estou falando besteira.

mariobaldini, 2017-03-29:
Você respondeu que é possível detectar a falha;

Mas ainda estou em duvida sobre: É possível detectar qual fonte de clock está sendo usada?

(que seria independente de detecção de falha). Caso não existir falha alguma, é possível eu verificar e confirmar qual fonte está sendo usada *no momento*? (e não qual foi configurada no setup)

eldertramontin, 2017-03-29:
Uma correção referente a minha terminologia: estávamos usando "falhas" porém olhei novamente no datasheet e percebi que ele detecta "faltas". Ou seja, não necessariamente vão ocorrer falhas para que esta detecção ocorra.

Atualmente estamos configurados assim:
MCLK <= XT2 / 2
SMCLK <= XT2 / 2
ACLK <= XT1

Onde:
XT1 é o cristal 1, com frequência 32768Hz;
XT2 é o cristal 2, com frequência 32MHz;
MCLK é o clock principal, responsável pela frequência de execução das instruções. Nesta configuração = 16MHz.
SMCLK é o clock para subsistemas, que pode ser usado como frequência de referência para UART/I2C/SPI/ADC/Timers. Nesta configuração = 16MHz.
SMCLK é o clock para subsistemas, que pode ser usado como frequência de referência para UART/I2C/SPI/ADC/Timers. Nesta configuração = 32768Hz e é utilizado como base de tempo.

Em caso de falta no XT1, clocks associados a ele usarão o REFO como fonte de clock. Em caso de falta do XT2, clocks associados a ele usarão o DCO como fonte de clock. REFO opera a 32768Hz.
Atualmente o DCO ainda não está configurado para trabalhar a 16MHz, mas será feito. Então, mesmo em caso de uma falta detectada, não haveria mudança na frequência de trabalho, somente na precisão. 

Agora respondendo a pergunta: É possível detectar qual fonte de clock está sendo usada?
Não exatamente. A partir do momento que for configurada uma fonte para determinado clock, se não houverem faltas, pode-se inferir que aquela fonte está sendo realmente usada. Se houver uma falta, a fonte de clock será alterada para DCO ou REFO, dependendo do cristal onde ela ocorreu. Então é possível estimar que determinada falta levou o sistema ao estado X ou Y, apenas lendo as _fault flags_. 

Para retornar ao estado original, os _fault flags_ devem ser resetados. Se a falta persistir, eles serão automaticamente setados. Caso contrário, as configurações originais, anteriores a falta, serão restauradas.

eldertramontin, 2017-04-05:
Os clocks redundantes foram configurados. Para checar as fault flags foi aberta a issue #70.

closed 2017-04-05 by eldertramontin