floripasat/obdh #68: PCB 2 - SPI1 Bus Bug

mfrata, 2017-03-26:
O canal da spi1 da placa 2 está apresentando problemas. 

Primeiramente havia um curto entre as linhas do barramento. Foi descoberto que este estava nos pinos do msp (A). Foi resoldado algumas vezes e por fim trocado por um msp(B).

O barramento não apresenta mais curto, porém a leitura com o cartão sd não funciona ainda. 

Este barramento é critico pois alem do cartão sd, as memorias e os painéis estão conectados à ele.

Portanto preciso que seja verificado a integridade deste barramento na placa 2. 

André e Arthur devem fazer testes usando uma launchpad para tentar se comunicar com o msp via spi. Devem observar as formas de onda no analizador lógico ou osciloscópio.

O André fará o código na launchpad e o Arthur(que ja esta fazendo um driver simples de spi) o ajudará no processo.

Se o problema for na pcb (trilhas do barramento) perderemos varias funções desta placa ou ela por inteiro.

PS: o msp(A) foi soldado na terceira placa junto com os circuitos do cartão sd, e não apresentaram problema nenhum. Portanto o problema não é interno ao chip ou ao cartão sd. E sim à placa 2.

mfrata, 2017-03-29:
Devido ao tempo que levara para encontrar o erro, e a incerteza de possibilidade de correção decidi deixar esta placa de lado.

Atualmente temos 
1 placa v2.0 com: 

1. msp
2. cristais
3. circuito de consumo de corrente total do obdh
4. circuito de referencia para o ADC
5. imu(9250)
6. sdcard
7. watchdog externo
8. 1 driver de corrente(magnetorquers)
9. 1 sensor de fotodiodo

1 versão 2.1 completa(exceto segunda imu(bosch))

1 versão 2.1 com o minimo soldado

mfrata, 2017-03-29:
Para essa PCB 3 - não soldaremos a segunda imu e nem os magnetorquers
Como essa placa será a da integração, não terá headers e nem devera ter algum fio soldado diretamente à pcb

mfrata, 2017-03-29:
Por fim ainda temos 1 placa da versão 2.0 que pode ser usada caso falte placas do obdh para os desenvolvedores do SW.

Totalizando 3 placas soldadas e um disponivel para solda.

mariobaldini, 2017-03-29:
Esse problema na SPI aconteceu na v2.1? 

Foi um problema de fabricação/montagem ou pode ser de layout? Com certeza a parte de SPI no layout 2.0 e 2.1 é idêntica?

mfrata, 2017-03-30:
Especificamente na placa 2. 

Muito provavel que tenha sido um problema de fabricação. Essas placas não tem qualidade muito boa. Vimos isso tanto no eps e ttc. 

Sim o SPI das duas versões são identicas.

mariobaldini, 2017-03-30:
"Especificamente na placa 2."

2.0 ou 2.1? 

Qual foi o fabricante?

mfrata, 2017-03-30:
2.1. 

A versão 2.0 foi pela curta
A versão 2.1 foi pela Griffus

closed 2017-04-04 by mfrata