floripasat/obdh #98: Implements a SPI interface through software

eldertramontin, 2017-06-21:
Creates a software that implements the SPI protocol over GPIOs without using a embedded peripheral SPI interface

mariobaldini, 2017-06-21:
Poderia descrever mais essa issue? E o motivo?

eldertramontin, 2017-06-21:
Precisamos de um canal de comunicação com a MCU do TT&C, porém não temos mais interfaces de comunicação livres (usamos 3 I²Cs, 2 SPIs e 1 UART). Já possuímos 4 GPIOs que ligam OBDH com TT&C, que já eram destinados para esta comunicação, porém sem um protocolo definido. Em vez de criar um protocolo novo, foi decidido utilizar o SPI, porém via software, para evitar mudanças no hardware entre modelo de engenharia v2 e modelo de vôo. Uma outra alternativa é utilizar o barramento I²C principal, onde já estão conectados EPS, OBDH e os 2 payloads. O receio é sobrecarregar o barramento, uma vez que outras equipes relataram problemas com I²C e colocar mais 1 nó no barramento aumenta as chances de ocorrer algum erro. Então, a opção atual é utilizar SPI, porém, se houver alguma dificuldade, podemos optar por utilizar o barramento I²C. Deu pra entender ou ficou confuso?

mariobaldini, 2017-06-21:
Onde está o diagrama de blocos dessa arquitetura nova? E da anterior também? Procurei eles aqui mas não achei.
Assim fica mais fácil pra entender o que foi necessário mudar (e implicou nesse novo requisito).

closed 2017-07-05 by eldertramontin